bw1129/PIDtoolbox #82: MatLab installation fails on Windows 11 - Error finding installer class

Sol1du2, 2022-05-11:
I am having some trouble installing MatLab via the provided installer, on Windows 11.

I tried moving the installer to a shorter and simpler path as suggested after googling the error (some users point to the possibility of the error being due to non ascii characters in the path) but to no avail. A screenshot of the error is below. It pops up pretty much immediately after starting the installer.

[image]

Sol1du2, 2022-05-11:
Nevermind it seems the issue is indeed non ascii characters but in the user folder, not specifically where the installer is. Changing my username folder to only ascii characters resolved the issue.

This might also not be exclusive to win11

bw1129, 2022-05-11:
How long ago did you grab v0.55? That issue was fixed recently and just re-uploaded to 0.55

Sol1du2, 2022-05-11:
> How long ago did you grab v0.55? That issue was fixed recently and just re-uploaded to 0.55

A few minutes before my original post. 

Hm, unfortunately after installing I can not start it. The splash screen comes up and then it just closes

bw1129, 2022-05-11:
Read the ‘README_PLEASE.txt’ very carefully

Sol1du2, 2022-05-11:
I did but I'm still failing to see why it doesn't start 😆

bw1129, 2022-05-11:
Find me on Facebook messenger and I’ll talk you through it

closed 2023-05-11 by bw1129