bw1129/PIDtoolbox #83: Step Response shows "insufficient data" all of a sudden

JinxFPV, 2022-06-10:
I've been doing the basement tune on an iflight Beast F7 board and got the step response curve and everything else just fine.  The last two days I get results in the log viewer but "Insufficient data" in the step response tool.  I have no clue what changed.

bw1129, 2022-06-10:
That’s not an issue with Ptb. It just means there’s not enough movement on that axis to extract reasonable data. Cruising around won’t produce sufficient data. Sometimes you’re lucky and it’ll extract a few curves but check the “n” (number of observations). This should be above 20-30

JinxFPV, 2022-06-10:
Thank you, I will check on it. But I was getting readings before using the same method - on angle mode and moving the sticks to their extremes for 15-20 sec.

bw1129, 2022-06-10:
Make sure you’re not using FF when examining step resp

JinxFPV, 2022-06-10:
Ah, that is probably it! I had forgotten about that and added it back in during the tuning process - thank you!

closed 2022-06-10 by JinxFPV