bw1129/PIDtoolbox #85: Spectrum Analyzer error on Linux (v0.55)

mluessi, 2022-06-12:
I'm using PIDtoolbox on Linux and everything seems to be working except the Spectrum Analyzer. When opening it, it plots the first spectrum but then the mouse pointer is indicating that it is hanging and I get the output below in the terminal from which PIDtoolbox was started.

The problem may due to the library that can't be reloaded, but I'm not sure why this happens. Any ideas?

```
Can't reload '/opt/MATLAB_Runtime/v93/bin/glnxa64/libmwcoder_types.so'

Error in finddelay (line 210)

Error in PTspec2DUIcontrol (line 190)

Error using waitforallfiguresclosed (line 9)
Error while evaluating UIControl Callback.

Index exceeds matrix dimensions.

Error in PTplotSpec2D (line 150) 
```

Canaill51, 2022-06-15:
> I'm using PIDtoolbox on Linux and everything seems to be working except the Spectrum Analyzer. When opening it, it plots the first spectrum but then the mouse pointer is indicating that it is [...]

Did you take a look at this https://github.com/bw1129/PIDtoolbox/issues/80?

On last posts, you could find your solution.

bw1129, 2022-10-11:
Someone posted a fix for that already. Look at the issues

closed 2023-05-11 by bw1129