bw1129/PIDtoolbox #86: PIDtoolbox v0.55 Error in PTload when selecting log (Linux)

abienias97, 2022-06-16:
Hello,

i just installed PIDtoolbox on Linux. Right after program open's up i want to select log in control panel on the right side. After selecting correct file a popup window from screenshot appears. I tried opening just this one log, unfortunately i don't have any other log to check right now. Zipped log file attached.

After clicking ok console says:

Error in PTload 

 Message:
Index exceeds matrix dimensions.
Index exceeds matrix dimensions.

Error in PTviewerUIcontrol (line 92)

Error using waitforallfiguresclosed (line 9)
Error while evaluating UIControl Callback.

After clicking ok in popup window, second popup with progress bar stays on at 0% and doesnt do anything.

Please help.

[image]
[log.zip](https://github.com/bw1129/PIDtoolbox/files/8919604/log.zip)

bw1129, 2022-06-16:
Make sure log is not in main

abienias97, 2022-06-16:
> Make sure log is not in main

It is clearly not inside main folder.

bw1129, 2022-06-16:
I have no issue loading that logfile on my end

abienias97, 2022-06-16:
Are you also using linux version?

abienias97, 2022-06-17:
I have no idea what has changed but it randomly started working just fine.

mituritsyn, 2022-08-14:
works fine on Fedora. @abienias97 could you please close the issue?

bw1129, 2022-08-14:
I recommend leaving it open so others can see the solution

mituritsyn, 2022-08-14:
> I recommend leaving it open so others can see the solution

there is no solution as there weren't issue) All issues also available after closing. The problem is there is so many irrelevant opened issues so it is hard to find relevant one

closed 2022-08-14 by bw1129