bw1129/PIDtoolbox #87: Error in PTload MacOS

mtshare, 2022-06-21:
Hi guys,

I'm trying to analyze my LOG from my quads but on both my Mac (iMac i7 & MacBook Pro M1) I'm getting this error (see attached image).
Can you help me figure out what's wrong?
This log works fine on Blackbox Explorer.

Attached you can find the ZIP with my BFL log.

Thank you so much.

[image]
[LOG00003.BFL.zip](https://github.com/bw1129/PIDtoolbox/files/8952371/LOG00003.BFL.zip)

bw1129, 2022-06-21:
no issues on my end. notice your table bottom right. You have not set things up correctly. Follow the 'README_PLEASE.txt.' file
[image]

bw1129, 2022-06-24:
Do NOT put logfiles in the main folder for v0.55

bw1129, 2022-06-24:
your first mistake is running the application from the apps folder. I stated not to do that.

mtshare, 2022-06-24:
Thank you for the support and sorry to bother you.
I really followed all the steps of the README file:

1. drop this 'main' folder and all of it's contents onto your desktop **DONE √**
2. Run the 'PIDtoolbox.app' program file directly from that 'main' **DONE √**

I still get this error:
[https://drive.google.com/file/d/13_3sg9wFGlNaX1BLYobryPA_mAtKRpCD/view?usp=sharing](url)

But it doesn't metter, my workatround is to put a copy of the log in the main folder and it works.

bw1129, 2022-06-24:
join zoom chat and I'll fix it up. https://queensu.zoom.us/j/95721856459?pwd=UUtiSTRUQUxBck5RSFMxcGwzOGZQQT09

closed 2023-05-11 by bw1129