bw1129/PIDtoolbox #88: Unable to open logs

Flyngd, 2022-07-19:
Hello , PT freeze when opening any BF log . The program was working perfectly a few months ago (last time I needed it) but now it wont open a single log . I tried to un/reinstall but it still the same . It can be with the "please wait message" forever . At the bottom right says: "Unable to set defaults".

Any help is apreciated

[image]

bw1129, 2022-07-19:
please make sure you're following the README_PLEASE.txt file

[Pidtoolbox error]<https://can01.safelinks.protection.outlook.com/?url=https%3A%2F%2Fuser-images.githubusercontent.com%2F33003938%2F179846371-3928ebc3-5e7b-4f87-ad75-346901cb259a.jpg&data=05%7C01%7Cbrian.white%40queensu.ca%7Cc07d8806bdc7463b780808da69cb98c8%7Cd61ecb3b38b142d582c4efb2838b925c%7C1%7C0%7C637938620401370050%7CUnknown%7CTWFpbGZsb3d8eyJWIjoiMC4wLjAwMDAiLCJQIjoiV2luMzIiLCJBTiI6Ik1haWwiLCJXVCI6Mn0%3D%7C3000%7C%7C%7C&sdata=zgp9nK%2B9EOTmZPUlCLtkw73lFobkAhYRjAGAYsonKUo%3D&reserved=0>

Flyngd, 2022-07-24:
Sorry about the delay on answer (holidays time) , I follow the readme.txt but unfortunately still dont open the logs . I tried to search on the Pc the "PTBdefaults" and "logfileDir" text files , but they are not any where on my computer . And it still sending the message "Unable to set user defaults".

The funny part is I have tried also on a laptop , it never had have PTbox installed on it , but it reproduces the exact same error .

Thanks for your help

bw1129, 2022-07-24:
If this is max, Go to ‘users/shared’ and delete the ‘PTB-…txt’ file then start again. Follow 2 main rules: 1) don’t put logfiles in main folder, 2) don’t run Ptb from application folder (use the version that’s in ‘main’ with that folder placed on your desktop. When asked via that pop up for the directory, navigate to that main folder. If you’ve done all that correctly it should work

Flyngd, 2022-07-24:
Finally it works, I moved all the folders to another hard disk and then it started to create new "default" and "log" files. Since then it works perfectly.
(Of course, it's a problem on my PC)

Thank you very much for your quick answer

AlexZ550Proton, 2022-08-15:
I have exact the same issue on Macbook Pro M1. The version of PIDToolbox is 2.57. The same version runs fine on Windows PC. However, on Mac the program stack on opening BB file. I run the app from main folder and BB id not under main (I tried different locations but no effect). Is anyone else experiencing the same issue?

bw1129, 2022-08-15:
Read the ‘readme_please.txt’

Message ID: ***@***.***>

Delete the ‘PTB…txt’ files in ‘users/shared’, Then start over following that readme as carefully as possible

AlexZ550Proton, 2022-08-15:
Got it. I had to select main folder at the first run. It works now. Thank you

closed 2023-05-11 by bw1129