bw1129/PIDtoolbox #91: MacOS M1 Installed and made a mistake on Step 4

xxchris82xx, 2022-09-01:
I've redirected the step 4 in the wrong folder.
How can I reset that and redirect to my Main folder?

bw1129, 2022-09-01:
go to ‘users/shared’ in the txt file ‘PTB…’ and either input the proper directory right in that text file then save it, or delete the file altogether and restart

Message ID: ***@***.***>

xxchris82xx, 2022-09-01:
Thanks, fixed!

kickingbird91, 2022-09-08:
I have this exact issue... this is when having read the directions would have come in handy. The only txt file I can find here is "read me" in the "main" folder. Would you mind elaborating on the fix for this issue please? I've looked elsewhere for a walkthrough of the fix for this issue but cannot find one. 

-Best

kickingbird91, 2022-09-08:
Never mind... I got it.