bw1129/PIDtoolbox #93: MacOS Ventura issues

nicesub, 2022-10-31:
Hello,

I have used in the past the 0.58 version without any issues on the Mac Air (one with old intel inside) on the older version of the OS and did not have any issues with the app. 
This night I upgraded the laptop to Ventura (latest version of OSx) and suddenly the app does not work anymore. It seems that when I select the blackbox file to load the app tries to process it but it struggles to bring forward the progress bar of the load. Also the file that I use for testing the app has two flights in it and before the app would ask me which flight to load. Now under the new OS the app struggles to bring that pop-up forward and I am unable to proceed with the analysis. Note, I am not a Mac user I just borrow this thing from my girlfriend when I go in the field to tune, the battery life is good and is portable thing. I don't know how to screenshot on it so I took the picture where you can see that the pop-up to select the flights is not showing up.
Link to the picture: [https://imgur.com/a/ZYMweOz](url)

 I have also deleted the PID tool box files and downloaded again with same issues. Again, this app version worked fine on the older OSX.[](url)

bw1129, 2022-10-31:
It’s unfortunately a no-go with Ventura until I upgrade my version of Matlab.

nicesub, 2022-10-31:
Ugh now I am stuck I suppose....
Looking forward for 0.59 I guess :)

Thnx for awesome tool, makes tunning verys simple and approachable

w453347000, 2023-01-11:
我的是M2的macbookair，也是Ventura的系统，我下载了055的版本，能打开软件但是加载黑盒数据后及卡死不动了，今天我下载了059，安装后打不开软件。。。右键点击图标显示有帮助对话框，但桌面上没有任何显示，看不到软件的任何界面

alber70g, 2023-01-23:
Is it possible to compile our own version with a newer version of Matlab?

sbladesltd, 2023-01-24:
do we have an ETA on when PID toolbox will be ready for working with Ventura OS?
thanks again for all your work!

roligmue, 2023-02-01:
Same here. Looking forward to support for Ventura OS support.

Thanks

amdeus, 2023-02-02:
Same here problems here. Keep up your awesome work:)

Verohomie, 2023-02-09:
Likewise... missing PTB on my old Mac. Might have to run on my wife's windoze 🥲

cdshepster, 2023-02-17:
> Is it possible to compile our own version with a newer version of Matlab?

Silicon Mac: New to Matlab, but after installing the IDE via trial, download mac osx source and open folder in MATLAB. Copy blackbox_decode from the PIDToolboxOSX download to the osx source folder. sudo chmod +x blackbox_decode. Run the PIDToolbox.m file and no issues running.

Verohomie, 2023-02-17:
> > Is it possible to compile our own version with a newer version of Matlab?
> 
> Silicon Mac: New to Matlab, but after installing the IDE via trial, download mac osx source and open folder in MATLAB. Copy blackbox_decode from the PIDToolboxOSX download to the osx source folder. sudo chmod +x blackbox_decode. Run the PIDToolbox.m file. The missing libraries will error in the log, select the hyperlink mentioned and install. Restart MATLAB and run.

FANTASTIC... please make a howto with links and terminal commands! TIA - BRadFPV

cdshepster, 2023-02-17:
> FANTASTIC... please make a howto with links and terminal commands! TIA - BRadFPV

Video on building locally - OSX Ventura: https://youtu.be/miylvMkRUok
Sign up for a matlab trial if you dont have an account and follow along.

nicesub, 2023-02-17:
But MATLAB is trial for 30 days, after days what then?

cdshepster, 2023-02-17:
> But MATLAB is trial for 30 days, after 30 days what then?

My hope is to fix the errors in that time and build into a permanent mac application to keep on my system as the fix 🤷🏻 idk though... still new to the application - will share those steps if I get that far... otherwise might be trying to create another trial next time I build a new quad.

Verohomie, 2023-02-22:
> Sign up for a matlab trial if you dont have an account and follow along. You will get some errors in the log window the first time running if you don't have the below packages installed.
> 
> You can select Add-On's from the Home Tab and install the required packages (resolves issues): -Curve Fitting Toolbox -Signal Processing Toolbox -Statistics and Machine Learning Toolbox

In addition to the toolboxes listed above add...
-Image Processing Toolbox

wildman199, 2023-04-04:
> Silicon Mac: New to Matlab, but after installing the IDE via trial, download mac osx source and open folder in MATLAB. Copy blackbox_decode from the PIDToolboxOSX download to the osx source folder. [...]

> Hello,
>
> I have used in the past the 0.58 version without any issues on the Mac Air (one with old intel inside) on the older version of the OS and did not have any issues with the app. This night I [...]

what a complete headache just to tune a drone.

wildman199, 2023-04-04:
what a complete headache just to tune a drone.

wildman199, 2023-04-04:
> Silicon Mac: New to Matlab, but after installing the IDE via trial, download mac osx source and open folder in MATLAB. Copy blackbox_decode from the PIDToolboxOSX download to the osx source folder. [...]

did this and now I get'PTlinecmap' is not found in the current folder or on the MATLAB
path, but exists in:

bw1129, 2023-04-04:
Yes you can run Matlab from the matlab command prompt if you have at least a trial version. But That’s not the same thing as compiling. PTB does not compile properly for Ventura and that is a Matlab problem. I’m looking into it and soon as Matlab fixes the issue then I can compile for distribution

WolfxFPV, 2023-04-09:
Hi
 I did all the things from this video https://www.youtube.com/watch?v=miylvMkRUok but the place with "select" doesn't show up 
[image]
Maybe I did something wrong but idk what to do

bw1129, 2023-04-09:
No go in Ventura yet

moorfpv, 2023-04-18:
This is really very sad. Have Y a timeline ?

Marceldu13, 2023-04-19:
Thank you for great job, No go for Ventura Yet means an option will be here ? Soon? THX

llwingok1, 2023-05-05:
> 嘿， 我做了这个视频中的所有事情https://www.youtube.com/watch?v=miylvMkRUok但是有“选择”的地方没有出现 也许我做错了什么但我不知道该怎么做做 [image]

重新装一下MATLAB，把这三个软件包选上，我是这样解决的。
-Curve Fitting Toolbox
-Signal Processing Toolbox
-Statistics and Machine Learning Toolbox

closed 2023-05-11 by bw1129