bw1129/PIDtoolbox #94: Trying to install PTB in windows 10 but not working

rokfpv, 2022-11-17:
I installed the PTB latest version in Desktop foler and Matlab automatically installed in program folder.
Afterward, I running the PTB in main folder but it won't open or any errors come.

DenchiG13, 2023-01-01:
I have the same problem

FlopBuster, 2023-02-26:
any solutions?

treadwm, 2023-03-13:
Same issue here.

FlopBuster, 2023-03-15:
solved by cleaning the %TEMP% folder.

treadwm, 2023-03-25:
Having similar issue on Win10. Cleaning the %TEMP% folder did not help. Uninstalled everything, rebooted, ran installer script, extracted files to folder on desktop and ran exe out of main. I see splash screen and it stops. No other obvious errors. All folders involved are lower case with no odd characters.

treadwm, 2023-03-30:
@NoSelph Unfortunately uninstall/reinstall didn't work for me. To be clear, this is on a laptop with a relatively new Win10 install. The only other things installed on it is prusa slicer and Betaflight configurator. It's only ever had the 0.59 version of PTB attempted to be installed. This same install procedure worked w/o issue on another pc of mine so I'm a bit confused.

delagra, 2023-04-06:
Same issue, installed, ran the PIDtoolbox.exe from main folder, splash screen loads and that's it. Cleaned up temp folder, uninstalled and reinstalled runtime, restarted the computer, tried various compatibility options, nothing helps.

closed 2023-05-11 by bw1129