bw1129/PIDtoolbox #104: PIDTtoolbox wont stop loading file

kylarstar123, 2023-03-07:
I got some betaflight logs i want open with the PIDtoolbox but it just says please wait... and wont open the file, no matter how long i wait.
Can somebody help me pls.

kylarstar123, 2023-03-07:
Forget it i found the

treadwm, 2023-03-13:
What was the cause? I am seeing the same thing loading BB files from a SpeedyBee fc.

treadwm, 2023-03-14:
Every file that gets opened generates a window w/o text that shows about 60% profgress.
[image]
Also, if I uncheck motors1-4, the cursor stops clocking and the graph draws normally. However, the null windows are still in the background until I close them.

bw1129, 2023-03-14:
Send the logfile.

treadwm, 2023-03-14:
[LOG00002.zip](https://github.com/bw1129/PIDtoolbox/files/10968423/LOG00002.zip)

bw1129, 2023-03-14:
Looked at logfile. Problem is you have logging off for several variables. No motors etc. set it up to properly log everything in BF configurator

treadwm, 2023-03-14:
Thx for confirming. I was not aware those could be turned off. Brand new FC w/ minimal presets added.
Where in BF are those enabled?

bw1129, 2023-03-14:
under cli, type get blackbox_disable to see full list:
# get blackbox_disable
blackbox_disable_pids = OFF
Allowed values: OFF, ON

blackbox_disable_rc = OFF
Allowed values: OFF, ON

blackbox_disable_setpoint = OFF
Allowed values: OFF, ON

blackbox_disable_bat = OFF
Allowed values: OFF, ON

blackbox_disable_mag = OFF
Allowed values: OFF, ON

blackbox_disable_alt = OFF
Allowed values: OFF, ON

blackbox_disable_rssi = OFF
Allowed values: OFF, ON

blackbox_disable_gyro = OFF
Allowed values: OFF, ON

blackbox_disable_acc = OFF
Allowed values: OFF, ON

blackbox_disable_debug = OFF
Allowed values: OFF, ON

blackbox_disable_motors = OFF
Allowed values: OFF, ON

blackbox_disable_gps = OFF
Allowed values: OFF, ON

treadwm, 2023-03-14:
Awesome! That solved it.
Thank you for your help w/ this.

ramonsaraiva, 2023-03-14:
I have a similar issue, but my loading doesn't ever get out of 0%.

[btfl_blackbox_logs.zip](https://github.com/bw1129/PIDtoolbox/files/10971529/btfl_blackbox_logs.zip)

These are my logs and the output of `get blackbox_disable` is:

```
blackbox_disable_pids = OFF
Allowed values: OFF, ON

blackbox_disable_rc = OFF
Allowed values: OFF, ON

blackbox_disable_setpoint = OFF
Allowed values: OFF, ON

blackbox_disable_bat = OFF
Allowed values: OFF, ON

blackbox_disable_mag = OFF
Allowed values: OFF, ON

blackbox_disable_alt = OFF
Allowed values: OFF, ON

blackbox_disable_rssi = OFF
Allowed values: OFF, ON

blackbox_disable_gyro = OFF
Allowed values: OFF, ON

blackbox_disable_acc = OFF
Allowed values: OFF, ON

blackbox_disable_debug = OFF
Allowed values: OFF, ON

blackbox_disable_motors = OFF
Allowed values: OFF, ON

blackbox_disable_gps = OFF
Allowed values: OFF, ON
```

Also this is my configuration in the blackbox tab:

[image]

I'm using the windows v0.59 release and also used the runtime installer.

bw1129, 2023-03-14:
Yes, the all caps doesn’t seem to work that well. I need BOLD, ALL CAPS, RED, FLASHING, BEEPING readme please 😂

ramonsaraiva, 2023-03-14:
Would it help add to https://github.com/bw1129/PIDtoolbox/blob/master/README.md instructions to read the `README_PLEASE.txt` file? Not sure if that's the ideal location but might help?

closed 2023-05-11 by bw1129