bw1129/PIDtoolbox #106: 0.59version's problem

Nicine, 2023-03-19:
my laptop is M1 Macbook,i found question that the first time i install pidtoolbox i could successfully use all functions such as filter analysis and pid curve analysis.But when i close the software,next time i open pidtoolbox i could not use further tools anymore,the windows don't show up which only a transparent windows in dock.I have to uninstall and install matlab and pidtoolbox again.How this issue can be solve？

junfpv, 2023-03-26:
I have encountered the same problem. After opening the black box file, no window can be opened. How can I solve it?

FlyFrg, 2023-03-28:
I have encountered the same problem.

karol-pilch, 2023-04-04:
I have installed version 0.59 (the first time I ever installed this) and no windows open apart from the main window for me.

Apple M1 Pro, MacOS 13.3.

crx100, 2023-04-04:
Ditto, Mac Air, and VenturaOS, only appears as transparent windows for the log viewer and spectral analyzer.

toni1809, 2023-04-06:
I have the same problem when instaling 0.59 verison I cant open it and only have invisible help dialoge screen.

Moavw, 2023-04-07:
I can't open files .log on any version on an intel mac

closed 2023-05-11 by bw1129