bw1129/PIDtoolbox #112: Missing "Gyro prefilt" from INAV 6.0.0 blackbox

slackista, 2023-04-08:
Due to changes in INAV 6.0.0, raw gyro data is not showing after "Gyro prefilt" is selected.
PR for that change https://github.com/iNavFlight/inav/pull/8139

bw1129, 2023-04-08:
Yes, I need to upgrade Ptb for inav 6. I’d be grateful for a logfile that contains prefiltered gyro so I can make changes for next version of Ptb.

slackista, 2023-04-08:
here you go :)
https://file.io/JxZ0Cpmnt8kK

bw1129, 2023-04-18:
@slackista you deleted the logfile. I didnt download it yet. I cant see throttle value in current INAV 6 log I have. no 4th column for set point or RC command throttle. 
can you please drop link to file again, and can you confirm whether having throttle in logfile is an option? weird it is not there by default

bw1129, 2023-04-18:
@slackista also motor sigs not on, and rcData not on. can you turn those on. all I need is a 10 sec hover

bw1129, 2023-04-19:
@slackista done. will be part of new release 
[image]

slackista, 2023-04-21:
When you expect to release new version?

bw1129, 2023-04-21:
It’s close. Couple of weeks maybe less

closed 2023-05-11 by bw1129