bw1129/PIDtoolbox #114: Selection of multiple flights from single file

fabi9163, 2023-05-01:
Hello @bw1129,
first: Thanks for this great software!

this time I've installed PIDtoolbox on Ubuntu like explained, works so far.
Problem: If I select a bbl-file I am not able to select multiple flights in this file like I can under Windows.
There is only a dialog in which I can input one Digit/Integer. 
Question: Is there some parsing eg. "1-4" to allow multiple selects? If not maybe you could add a "all" option on index 0.

Thanks Fabi

closed 2023-05-11 by bw1129

fabi9163, 2023-06-20:
@bw1129 The last Linux-Version seems to be released before my issue-report on 12 April.
You closed the Issue with "completed" on 11 May.

Question: Is there a way/howto to run/compile on Linux or was this a mistake?

Thanks for your answer

bw1129, 2023-06-20:
Sorry, no go