bw1129/PIDtoolbox #116: Can't get log files to load and be analyzed on MacOS Ventura - v0.6

chrissetty, 2023-05-11:
I installed the new 0.6 version on my Mac running Venture and it got much, much further than before, however when I try and load a log file, it just sits there with a window that has a loading bar at about 75% and says "please wait" - no matter how long I let it sit there, it never progresses or loads any of the log files. I have placed the app folder on the Desktop, run it from there, set the Main Working Directory correctly and ensured all the log files are unlocked. Still won't load any log files.
[image]

bw1129, 2023-05-11:
Go to ‘main’ folder and send me the ‘PTBlog’ file

chrissetty, 2023-05-11:
Hmmmmm….this is what is in the main directory…don’t see a PTBlog file

￼

bw1129, 2023-05-11:
if you do not know where the 'main' folder is then you did not follow the installation instructions.

chrissetty, 2023-05-11:
Ok….Isn’t this the Main folder…that is what the name of the directory is and also where the application is, per the instructions in the readme

￼

closed 2023-05-11 by bw1129