bw1129/PIDtoolbox #117: PIDTool box not loading files

b4tn, 2023-05-13:
I get a dialog box that looks like it should have some buttons but nothing is clickable. The progress arrow just hangs. Using betaflight 4.4.1 and ptb .61. I tried removing all versions of ptb and maylab and doing a fresh install. Tried multiple different logs as well. 
[image]

bw1129, 2023-05-13:
Send logfile and send PTBlog.txt found in the ‘main’ folder

b4tn, 2023-05-13:
added, thanks
[PTBlog.txt](https://github.com/bw1129/PIDtoolbox/files/11470797/PTBlog.txt)
[btfl_006.txt](https://github.com/bw1129/PIDtoolbox/files/11470798/btfl_006.txt)

bw1129, 2023-05-13:
this means the logfile is missing some critical information. you have set logging OFF for set point and possibly other variables that are needed to run PTB. 
[image]

bw1129, 2023-05-13:
the only fields available in the logfile are these. gyroADC is filtered gyro, plus you have debug, but no set point, no PID terms, no prefiltered gyro. 
[image]

bw1129, 2023-05-13:
so this is not an issue with PIDtoolbox, it is an issue with how you are logging in Betaflight. I will close this issue.

closed 2023-05-13 by bw1129

b4tn, 2023-05-14:
It’s a new build. There are not many options in BF for log setup. It has default tune, logging rate set to 1/2 (4khz) and debug is gyro scaled. FC is a mamba MK4.

b4tn, 2023-05-16:
If anyone stubles across this I found the issue.  In Betaflight presets there is a Gyroflow preset that reduces logging to only whats needed for processing Gyroflow video.  This preset breaks the logging function for PID toolbox.