bw1129/PIDtoolbox #118: INAV 6.1 blackbox data not loading

sdellava, 2023-05-15:
I've downloaded the last PIDtoolbox version and tried to load data collected by a drone running INAV 6.1.

The "please wait" popup appear without any other information. The load never end.

sdellava, 2023-05-16:
I've found no log. This is the INAV bb file that lead to the issue: 
[blackbox_log_2023-05-15_141409.zip](https://github.com/bw1129/PIDtoolbox/files/11486066/blackbox_log_2023-05-15_141409.zip)

bw1129, 2023-05-16:
Both INAV and BF have options on what parameters to log. For both you much log set point and gyro or PTB will not run. This is done in CLI for BF. For INAV 6+ it is found in the gui. Please make sure you are logging set point, gyro, gyro raw (inav) and gyro_scaled (BF)

bw1129, 2023-05-22:
Make sure you are logging set point. Otherwise it will crash when loading.

bw1129, 2023-05-22:
Also if you don’t see a PTBlog.txt file you probably did not follow the read me.

sdellava, 2023-05-22:
I've uninstalled, dowloaded again and run as stand alone. This way it works.

So I suppose the installation is not done correctly.

bw1129, 2023-05-22:
Just tried your logfile. It works fine on my end using v0.61  
[image]

bw1129, 2023-05-22:
this is not an issue with the software so I am closing the issue

closed 2023-05-22 by bw1129