bw1129/PIDtoolbox #120: Not able to load bbl file

RollingGecko, 2023-05-22:
I get an endless spinner when I try to load the log file. It is a bbl file from betaflight 4.4.1 with 2khz and scaled Gyro.

No idea what is going wrong. Tryed it on 2 computers. Both fresh installations with the latest Version of PIDtoolbox. 
[PI_0_7.zip](https://github.com/bw1129/PIDtoolbox/files/11537368/PI_0_7.zip)

bw1129, 2023-05-22:
I tried your logfile and it works fine on v0.61, so I am closing the issue. 
[image]

closed 2023-05-22 by bw1129

RollingGecko, 2023-05-24:
Thanks. Now I know at least, that the files are not corrupted. But I still have now I idea where the problem on my Windows System is. On Win11 and Win10.