bw1129/PIDtoolbox #125: How to compile to Binary on linux?

SnakebiteEF2000, 2023-07-26:
Hello, I have no expirience with matlab but i would like to use this tool on linux. Does someone know how i can build it from source?

ask6483, 2023-08-21:
Unfortunately you will need a copy of Matlab for Linux.

happenpappen, 2023-10-03:
I would love to see this on Linux as well - do you k now, if a trial version of Matlab will be ok to build the binary? Would you provide the steps how to build the binary?

happenpappen, 2023-10-08:
Would be nice, if @ask6483 or @bw1129 would have time to answer my question...

bw1129, 2023-10-08:
@happenpappen no Linux releases anymore, sorry.

dgsardina, 2023-11-16:
Dear @bw1129 , in [here](https://www.mathworks.com/matlabcentral/answers/573781-how-to-generate-a-standalone-application-for-linux) they suggest using the platform independent JAVA archive output. Do you think you could give it a try? I would happily test it.

Alternatively, if you can give us a hint on how to produce the linux binary we could do it ourselves. We can even create a cloned repo for the linux version only so people do not bother you with it.

bw1129, 2023-11-16:
As I said several times before, I no longer release source code anyway so there’s nothing to compile

closed 2023-11-16 by bw1129