bw1129/PIDtoolbox #126: Unable to function properly in the environment of MacOS 13.5 Apple Silicon M1

Trojan1986, 2023-07-29:
Data analysis cannot be performed after loading the black box file.

007kh12ro, 2023-08-09:
The same issue. "Please wait..." and nothing happen. But i can open resp tool with nothing

[image]

[image]

bw1129, 2023-08-09:
Pls Send the ‘PTBlog.txt’ file found in the ‘main’ folder

zvonler, 2023-09-26:
Just upgraded a 2020 M1 macbook to MacOS 13.6 and confirmed that a fresh install of PIDtoolbox 0.64 seems to work fine for loading blackbox files and running the step response tool. Are there other tests I could run to see if this issue is fixed?

danilocecilia, 2023-09-28:
Same problem for me, please wait pop up seems to get stuck there and can't do anything else.

Opened the log file and theres's no much details, only this path "/Users/daniloc/Desktop/PIDtoolbox_v0.64_osx/main/"

bw1129, 2023-09-28:
Please send the ‘PTBlog.txt’ file

Iamyahodfpv, 2023-10-25:
Im having the same issue and cannot find the 'PTBlog.txt' file

bw1129, 2023-10-25:
If you can’t find PTBlog.txt you are most likely not following the directions laid out in the README_PLEASE.txt file found in the download

Iamyahodfpv, 2023-10-25:
I followed the directions, not sure what I could've done wrong

Iamyahodfpv, 2023-10-25:
when running the installer what should I set as the destination folders?

bw1129, 2023-10-25:
Let runtime install where it wants to in the applications folder. But when running PTB you run the program provided, the one found in the main folder

Message ID: ***@***.***>

Iamyahodfpv, 2023-10-26:
I got it to work once but now it won't work again not sure why but it won't load the log files

bw1129, 2023-10-26:
Click the ‘reset main directory’ button and make sure you’re directing Ptb to the main folder where the program resides. You might have this going to the wrong location. You can see the directory in a file called ‘mainDir-PTBv0.6..’ found under ‘users/shared’

j03m4r13, 2023-12-15:
I got the same issue but I solved it by simply copying ALL the files inside the main and moving them outside of that folder. So let say the runtime folder is initially located at /Users/myname/Desktop/PIDtoolbox_v0.65_osx/main then I simply move those inside the main to  /Users/myname/Desktop/PIDtoolbox_v0.65_osx and that loaded the logfiles immediately.