bw1129/PIDtoolbox #127: MacOS - can't change current folder (name nonexistent)

vaska1992, 2023-10-10:
I followed the guide to set the current folder to one on my desktop, but PTB was in a subfolder on my desktop and apparently i had spaces in the folder name, and it seems to hate that, on the error message it shows my folder name without spaces, so i figured it doesnt like it and i cannot change it to choose another one, the application closes as soon as i click "ok" on the error message. 
How can i clean it up and do fresh install?

bw1129, 2023-10-10:
go to 'users/shared' and remove all txt files named 'mainDir-PTBv...' then try again

vaska1992, 2023-10-10:
Thanks! that did it. 
moral of the story: don't use spaces in folder path names :)

closed 2023-10-10 by vaska1992