bw1129/PIDtoolbox #128: PTB hangs on loading BBL on win10 ver064

HugCat, 2023-10-12:
PTB refuses to load many of my bbl files.
It hangs on loading about 30%. 
This problem  confuse me many days .TKS
[bbl.zip](https://github.com/bw1129/PIDtoolbox/files/12885427/bbl.zip)
[image]

HugCat, 2023-10-12:
earlier I use 0.59 suddnely can't load so i upgread to 0.64 but still not working
[PTBlog.txt](https://github.com/bw1129/PIDtoolbox/files/12885477/PTBlog.txt)

mangorille, 2023-10-15:
Have this problem as well. PTBlog.txt shows errors:

Index exceeds the number of array elements (0).

Error in PTviewerUIcontrol (line 70)

Error using waitforallfiguresclosed (line 9)
Error while evaluating UIControl Callback.

ishampadron, 2023-10-16:
I have this issue as well. Fresh install of PIDtoolbox 0.64. It just hangs on load. My logs open up fine in PIDAnalyzer.

mangorille, 2023-10-16:
Are you getting this error? Mine seems okay now:

https://github.com/bw1129/PIDtoolbox/issues/130

HugCat, 2023-10-18:
TKS,guys still not working,trying learn to use PIDAnalyzer

bw1129, 2023-10-18:
Join the PTB discord. Folks there can help you get it running. You can even DM me there and I’ll help directly

druckgott, 2023-11-13:
I have this error:
Index exceeds the number of array elements (0).

Error in PTviewerUIcontrol (line 70)

Error using waitforallfiguresclosed (line 9)
Error while evaluating UIControl Callback.

maxledfreak, 2023-11-14:
Hi,

get same behaviour during tuning my iNav quad. Looks like that this occurs after selecting a second file for analysis.

bw1129, 2023-11-14:
Try v0.65

druckgott, 2023-11-14:
I already used 0.65 and get the error. Can I Support in any way?

bw1129, 2023-11-14:
Windows or Mac?
send logfile and send PTBlog.txt file (in main folder)

druckgott, 2023-11-14:
Windows.
https://github.com/bw1129/PIDtoolbox/files/13336661/PTBlog.txt

miralex773, 2023-11-15:
With INAV PTB would freese on file import with above symptoms when gyro_raw is enabled. Without gyro_raw import is OK.

druckgott, 2023-11-15:
Ah ok yes i have also Switches on gyro raw so i have all things on what you can select

apeelme, 2023-12-31:
Same problem for me with 0.59 and 0.65 also.

mangorille, 2024-01-07:
I was able to resolve this by making sure to delete all the blackbox log files on the flight controllers and then starting from an empty flash.

surfdado, 2024-02-11:
v0.65 also hangs for us with bbl files (Betaflight 4.5), For now we downgraded to v0.59 to get around this issue.

bw1129, 2024-02-11:
almost always those errors happen because you didn’t follow the “readme please” instructions closely.