bw1129/PIDtoolbox #130: Index exceeds the number of array elements

mangorille, 2023-10-15:
On v0.64, cannot import Betaflight log files. 

When attempting to import, the import hangs after clicking Select all and Ok from the "Select file(s):" box that shows the duration of flights found in the log.

PTBlogs.txt shows errors at the end:

Index exceeds the number of array elements (0).

Error in PTviewerUIcontrol (line 70)

Error using waitforallfiguresclosed (line 9)
Error while evaluating UIControl Callback.

bw1129, 2023-10-15:
Yes I think there’s a bug when selecting all files

mangorille, 2023-10-15:
Tried selecting just one file, same error.

PtbVersion = 'v0.64'
CompType ='PCWIN64'
Matlab_Version ='9.5.0.944444 (R2018b)'

bw1129, 2023-10-16:
You have a different issue if you can’t run a single logfile. Note, it’s not going to work on BF development builds above 4.4.2

mangorille, 2023-10-16:
Thanks for the quick replies. Looks to be working okay now even for multiple files. Not sure if I just had a corrupt log file or if any of these changes are what fixed it:

Changed Sample Rate from 1KHz to 2 KHz
Changed from Gyro_Raw to Gyro_Scaled

closed 2023-10-16 by mangorille