bw1129/PIDtoolbox #131: PTB runs randomly in Windows 10

demvlad, 2023-12-16:
PTB 0.65 runs randomly in Windows 10. 
Often i can not run the program. In this case log file contains information:
Undefined function or variable 'ctfroot'.
Error in checkRequiredMCRProducts (line 22).
When PTB runs successfull then all work fine!
I have reinstalled programm many time, but it does not have some stable result.

demvlad, 2024-01-19:
I understand, bug like this is hardly to find..
But your program is really nice! Is it posible to run it directly from matlab/simulink from source code?

bw1129, 2024-03-09:
this is an indication you are running the wrong executable and not following the 'README_PLEASE.txt' doc.
please confirm

demvlad, 2024-03-09:
> this is an indication you are running the wrong executable and not following the 'README_PLEASE.txt' doc. please confirm

Thank you for answer.
I am following this file:
[README_PLEASE.txt](https://github.com/bw1129/PIDtoolbox/files/14546801/README_PLEASE.txt)

demvlad, 2024-03-11:
Seems, this is helpfull, i've tried to do this 3 time and had success:
https://www.mathworks.com/matlabcentral/answers/437462-why-am-i-not-able-to-execute-my-standalone-application-on-only-certain-end-user-machines

bw1129, 2024-03-30:
**SOLUTION:**

basically, if you open the 'PTBlof.txt' file and see the following
[image]

go to:
C:\Users\[username]\AppData\Local\Temp\[username]\mcrCache[version]
and delete all folders labeled 'mrcCache'

details can be found here:
https://www.mathworks.com/matlabcentral/answers/437462-why-am-i-not-able-to-execute-my-standalone-application-on-only-certain-end-user-machines

bw1129, 2024-03-30:
solution:
https://www.youtube.com/watch?v=0aGg9WlEdnQ