bw1129/PIDtoolbox #133: PID TOOL BOX cannot be opened at all since the MATLAB error response

fofkjho78w3hrf7835, 2024-01-11:
the tool box cannot be opened at all and the PTB log shows the error of the 22nd line

bw1129, 2024-03-09:
you have to provide more info. also make sure you are following closely the README_PLEASE.txt file

qvasic, 2024-03-27:
I believe I am having the same error. I installed and re-installed PIDtoolbox as described in README_PLEASE.txt, rebooted my PC multiple times in the process. I am running Windows 11 Pro build 22631.3296. I tried installing and running PIDtoolbox_v0.65_win. PTBlog.txt says:

```
Undefined function or variable 'ctfroot'.

Error in checkRequiredMCRProducts (line 22)
```

qvasic, 2024-03-27:
I downloaded 0.64 version, and it ran successfully without re-installation.

qvasic, 2024-03-28:
Aaaand the problem is back with 0.64. I didn't do anything, just viewed a bunch of black box logs.

bw1129, 2024-03-30:
the answer to this issue can be found in issue [#131 ](https://github.com/bw1129/PIDtoolbox/issues/131)
basically, if you open the 'PTBlof.txt' file and see the following
[image]

go to:
C:\Users\[username]\AppData\Local\Temp\[username]\mcrCache[version]
and delete all folders labeled 'mrcCache'

details can be found here:
https://www.mathworks.com/matlabcentral/answers/437462-why-am-i-not-able-to-execute-my-standalone-application-on-only-certain-end-user-machines

bw1129, 2024-03-30:
solution
https://www.youtube.com/watch?v=0aGg9WlEdnQ

user11459, 2024-04-05:
I cant open it.It said could not find MATLAB runtime.

bw1129, 2024-04-05:
@user11459 you have to follow the install instructions