bw1129/PIDtoolbox #135: Feature request

druckgott, 2024-03-24:
In Inav it´s possible to do in flight PID Tuning.

So it would be cool if it´s posible to select only a area of the log file to get analyzed by the tool. Then it´s possible to select the area where the PID values are stable. You can do one flight with differnte value and analyse the log file. Currently you have to land several times.
This is a example file where different pid values are tuned in one file:
[LOG00045.TXT](https://github.com/bw1129/PIDtoolbox/files/14734865/LOG00045.TXT)

[image]

bw1129, 2024-03-24:
PTB can already do this. But how do you get INAV to change PIDs in flight? That’s a question for INAV

druckgott, 2024-03-24:
Ok how does this work? I have already attached my log with Different pid values.

I did changed the pids like here described
https://youtu.be/1NRG0vmhzSI?si=A90SKZSQ8X9xDTM8

This works perfekt. But if i load the log file its one big file so it do Not detect the Different pids. I have t select the Area by any other way. Where can I find how to select only a Part of the log file?

bw1129, 2024-03-24:
Load the same logfile in multiple times then set difference trim windows on each to capture the time with each different pid settings. But note, if there is not constant input on the axis of interest it will give useless results. Also it’s not clear it’ll work at all on a wing anyway

bw1129, 2024-03-25:
@druckgott I just checked your logfile. The problem is PTB was never designed to work with fixed wing, so it expected at least 4 motors, but only one exists and it crashes. You did not mention that the program crashes. This is an easy fix. Here is a screenshot of your logfile 
[image]

bw1129, 2024-03-25:
@druckgott here is a screenshot of the step response for the data on roll 
[image]

bw1129, 2024-03-25:
@druckgott an example move on roll 
[image]

druckgott, 2024-03-25:
Is there any documentatio how to click for trimmung? I have checke it But sometimes it do Not work

druckgott, 2024-03-30:
Ok it works it would be realy cool to have the information of the events in a extra gaph or maybe as vertical lines in the Overview.