bw1129/PIDtoolbox #136: Windows 11 PIDtoolbox not starting past splash screen.

keiner0815, 2024-04-02:
I installed the program

keiner0815, 2024-04-02:
I found a PTBlog.txt at the desktop root whith following content:

____________________________________________________________________________________________________________________
PtbVersion =

    'v0.65'

CompType =

    'PCWIN64'

Matlab_Version =

    '9.5.0.944444 (R2018b)'

currentDate = 

  datetime

   2023-11-13

defaults =

  36x2 table

           Parameters              Values   
    _________________________    ___________

    'firmware'                   '1'        
    'LogViewer-SinglePanel'      '0'        
    'LogViewer-plotR'            '1'        
    'LogViewer-plotP'            '1'        
    'LogViewer-plotY'            '1'        
    'LogViewer-lineSmooth'       '1'        
    'LogViewer-lineWidth'        '3'        
    'LogViewer-Ymax'             '500'      
    'spec2D-term1'               '1'        
    'spec2D-term2'               '2'        
    'spec2D-smoothing'           '3'        
    'spec2D-delay'               '1'        
    'spec2D-plotR'               '1'        
    'spec2D-plotP'               '1'        
    'spec2D-plotY'               '1'        
    'spec2D-SinglePanel'         '0'        
    'FreqXthr-Column1'           '3'        
    'FreqXthr-Column2'           '2'        
    'FreqXthr-Column3'           '8'        
    'FreqXthr-Column4'           '7'        
    'FreqXthr-Preset'            '1'        
    'FreqXthr-Colormap'          '3'        
    'FreqXthr-Smoothing'         '3'        
    'FreqxTime-Preset'           '2'        
    'FreqxTime-FreqSmoothing'    '2'        
    'FreqxTime-TimeSmoothing'    '2'        
    'FreqxTime-Colormap'         '3'        
    'StepResp-plotR'             '1'        
    'StepResp-plotP'             '1'        
    'StepResp-plotY'             '1'        
    'StepResp-SinglePanel'       '0'        
    'StepResp-Ymax'              '1.75'     
    'Font-name'                  'Helvetica'
    'Font-size'                  '12'       
    'Font-weight'                'normal'   
    'Font-angle'                 'normal'   

Index exceeds the number of array elements (0).

Error in PTviewerUIcontrol (line 70)

Error using waitforallfiguresclosed (line 9)
Error while evaluating UIControl Callback.

____________________________________________________________________________________________________________________

Seems to be from another try to get the program to work.

I uninstalled everything including the left over folders and the Matlab Runtime. Also tried a newer version of Matlab runtime. 
Ended up uninstalling everything again and trying a new approach.

But ended up with the same result: If i open the program from the main folder I see the splash screen a few seconds, then it closes without any error message and without anyting in the PTBlog.txt file

keiner0815, 2024-04-10:
Thanks for your reply.
I tried to delete this folder. 

It had the effect that I see the splash screen for a longer time until it disappeared. But it did not load the programm either.

 I will have a look into the discord Server.

Canaill51, 2024-04-10:
Another question : Did you follow the Readme file when installing and using
PTB? Where do you run PTB from? Can you show how is it installed?

keiner0815, 2024-04-13:
I have these folders in C:\Programs
[image]

The
PIDtoolbox_v0.65_win
Folder was previously on the desktop as stated in the readme.
And I tried to start it witch the PIDtoolbox.exe
from this path (on the desktop)
"PIDtoolbox_v0.65_win\PIDtoolbox_v0.65_win\main"

Also tried to start it from this path:
C:\Program Files\PIDtoolbox\application

But both did not work.

keiner0815, 2024-04-13:
I tried to accept the invitation to discord, but I think I was too slow and the invitation has timed out as I get an error when trying to accept it.

Canaill51, 2024-04-13:
https://discord.gg/zeQUhj5n  (will expire in 7 days, hurry up lol)

chimera15, 2024-05-20:
same I get stuck when i try to open a log